ThunderFly-aerospace/TF-R1 #4: Zamlžování kamer

kaklik, 2018-11-23:
Do palubních kamer na střeše zřejmě prosakuje voda protože se zamlžují. 
Řešení asi je do obalu vhodně udělat odvětrávací dírku, aby případná vlhkost měla kudy uniknout z obalu kamery.

kaklik, 2018-12-03:
@roman-dvorak Zjistil že u IP kamer lze "zapnout vyhřívání".

roman-dvorak, 2018-12-03:
Chtělo by to zjistit, jestli tuto funkci mají obě kamery. V nastavení (přes webové rozhraní) je to pojmenované jako 'Defrosting" a lze tomu nastavit nějaká hodnota 0-100. V zapnutém stavu je to docela znát.

kaklik, 2018-12-10:
@roman-dvorak To musíme vyzkoušet na té kameře na autě, teď k ní není přístup. Jak se do toho webového rozhraní kamery dostaneš bez windows a IE?

roman-dvorak, 2018-12-11:
@kaklik Ano, musel jsem si znovu-rozfungovat windows a z IE s nějakým pluginem jsem se do toho rozhraní dostal. To můžeme vyzkoušet na kameře na autě.

kaklik, 2019-01-22:
Vyřešeno vysušením kamery a  vložením pytlíku se silicagelem.

closed 2019-01-22 by kaklik

roman-dvorak, 2019-10-06:
Zadní kamera se znovu zamlžovala. Byla na nějkou dobu otevřena a vyvětrána. Bylo znát, že vyhřívání funguje. Což zřejme způsobilo vypaření vlhkosti a zkondenzování na vnitřní straně průhledné koupulky.

kaklik, 2019-10-06:
Možná by bylo řešením kamery nechat trvale připojené a tím i zahřáté na provozní teplotu.