ThunderFly-aerospace/TF-R1 #5: Dochází k výpadkům záznamu

kaklik, 2018-11-23:
Současné řěšení záznamu je citlivé  na vnější vlivy. V případě že dojde k přerušění spojení k notebooku operátora, Tak se záznam přeruši. 

Zřejmě je potřeba pro záznam zavést samostatný počítač pevně spojený s TF-R1. Operátor by pak měl vidět pouze stream, který ale primárně neslouží k záznamu.

kaklik, 2018-12-10:
Tohle se zřejmě vyřeší dokončením úkolů https://github.com/ThunderFly-aerospace/Hardware/issues/12 a https://github.com/ThunderFly-aerospace/TF-R1/issues/3

kaklik, 2019-02-04:
Až na zotavení z krátkodobého výpadku spojení s kamerami je tento úkol zřejmě vyřešený.

roman-dvorak, 2019-06-27:
na VLC wiki stránce píšou poznámku o nutnosti použití správného kodeku pro nahrávání. Tedy to, co zjistil @JanKott 

[#Save_a_stream_with_VLC](https://wiki.videolan.org/Documentation:Streaming_HowTo/Receive_and_Save_a_Stream/#Save_a_stream_with_VLC)

Při spuštění nahrávání pomocí červeného tlačítka by se v terminálu mělo objevit automaticky získané optimální nastavení.

To lze pak použít do nahrávacích skriptů. Nahrávací skripty mají především výhodu v tom, že se snáze pozná, že kamery něco nahrávají a je možné tento záznam spouštět např. přes SSH.

roman-dvorak, 2019-06-30:
Postup popsaný na VLC wiki zafungoval. Nyní máme skripty, ze kterých vzniká použitelný záznam.

roman-dvorak, 2019-07-15:
Je potřeba upravit nahrávací skript (stejným postupem jako minule) tak, aby nahrával i zvuk z RTSP streamu.

roman-dvorak, 2019-09-01:
Na zadní kameře je nyní zaznamenán i zvuk.

closed 2019-09-01 by roman-dvorak