ThunderFly-aerospace/TF-R1 #7: Instalace palubního počítače

kaklik, 2018-12-10:
@ChroustJan  pořídí [palubní počítač](https://www.levne-notebooky-pc.cz/black-friday/profesionalni-notebook-s-core-i5-lenovo-thinkpad-t430) s [dokovací stanicí](https://www.superlevnapc.cz/ibm-4337-lenovo-thinkpad-mini-dock-series-3-usb-3-0-vcetne-klicu.html). 

Bude na něj potřeba nainstalovat OS a potřebné nástroje: 

- Ubuntu 18.04 (Desktop), naistaluje @kaklik paralelně se zachováním naistlavoných windows
- ROS nainstaluje @roman-dvorak 
- Qgroudcontrol server pro připojení QGC klienta - zde je trochu komplikace neboť zatím není jasné, jak lze vytvořit telemetrický server.

roman-dvorak, 2018-12-11:
Jako QGroundControl server by měla sloužit utilita `mavproxy.py`,  která se připojí na sériovku a vytváří UDP rozhraní. Zatím mi nefunguje úplně dobře. Po spuštění QGC se k mavproxy neustále připojuje a odpojue. Mavproxy vypadá připojené stále....

Zkusil jsem napsat, na jejich fórum jaké je nejlepší řešení. Tak počkám na odpovědi.

roman-dvorak, 2018-12-11:
Na foru mi bylo doporučeno toto: [](https://github.com/intel/mavlink-router). Po prvním otestování lokálně na PC se zdá, že by to mohlo fungovat. Instalace proběhla podle postupu na GitHubu bez problému.

Spuštění pak pomocí příkazu
`mavlink-routerd -e 0.0.0.0:14550 /dev/ttyACM1`

roman-dvorak, 2019-06-06:
V autě bylo nějakou dobu experimentováno s přenosem MAVLINKU přes MAVROS. Mělo to velmi špatnou spolehlivost a bylo to extrémně snadné rozbít - někdy ani nevím jak. Proto navrhuji pokusit se vrátit k mavlink routeru, který bude Mavlink z modemu překládat na UDP.  To budou poslouchat jednotliví klienti.

kaklik, 2019-06-16:
Routování mavlinku funguje.

closed 2019-06-16 by kaklik