ThunderFly-aerospace/TF-R1 #9: Indikátor správné rychlosti pro řidiče

kaklik, 2018-12-16:
Protože se startovní a přistávací pozemní rychlost mění podle rychlosti větru, teploty a atmosférického tlaku, je vhodné řidiči správnou rychlost indikovat. 
Indikátor může být buď jednoduchý se třemi barevnými/trojbarevnou LED. (Modrá, zelená, Červená). 
Kdy modrá led indikuje příliš nízkou rychlost, zelená optimální a červená příliš vysokou rychlost. 

Indikátor může být i složitější v nějaké formě HUD displeje, kde je pak možné zobrazovat i další informace, jako například optimální směr. 

Ve všech případech by indikátor měl obsahovat i heartbeat palubních naváděcích systémů automobilu.

roman-dvorak, 2019-06-06:
Jako vhodné řešení se zdá, aby palubní systém vybrané údaje (např airspeed) předčítal nahlas. To bude mít vlastnost, že tuto hodnotu uslyší jak řidič auta, který bude mít úkol držet určitou airspeed, tak i pilot. Pro toho airspeed je také užitečná, aby například instruoval řidiče, ze má nějakým způsobem zmenit rychlost.

roman-dvorak, 2019-06-12:
Je připravený skript, který pomocí TTS systému čte rychlost z airspeed senzoru z mavlinku. Routování probíhá přes #21

closed 2019-06-15 by roman-dvorak