ThunderFly-aerospace/TF-R1 #10: Cedulky na dokumentaci trosek

kaklik, 2019-03-18:
Do vybavení auta by bylo vhodné přidat vhodné cedulky pro označování trosek. 
Konstrukce cedulek by měla umožnit jejich umístění v různém terénu. Zároveň by cedulky měly být dobře viditelné, aby na ně někdo při hledání dalších trosek nešlápl. 
Identifikace cedulky by měla být z fotografie dobře čitelná.

cesilko1, 2019-03-19:
Ty cedulky mají mít svoje číslo už v 3D modelu, nebo se dodělá až na vytištěném modelu?

kaklik, 2019-03-19:
Záleží, jakým způsobem by se mělo na modelu pak dodělávat. Cílem je mít s tím co nejmíň práce.

kaklik, 2019-03-21:
Vidím, že jsi navrhunul tohle: 
[image]

Jak se to ale použije na asfaltové dráze? A jak se označí jednotlivé díly? 

[image]

closed 2019-04-21 by cesilko1