ThunderFly-aerospace/TF-R1 #11: Držák venkovní 12V zásuvky na startovací desku auta

kaklik, 2019-03-20:
Na  desku, která je přišroubovaná k nosiči startovacího auta by bylo potřeba připevnit několik zásuvek ze spodní strany.

Výkres zásuvky včetně rozměrů je k dispozici [tady](https://www.tme.eu/cz/details/procar-57608003/automobilove-napajeci-prislusenstvi/pro-car/57608003/).

Problém zásuvky je v tom, že z jedné strany není zakrytá a může do ní snadno natéct voda, což je nechtěná situace. Proto je potřeba mezi desku a zásuvku vymodelovat tisknoutelnou podložku, která bude sešroubovaná se zásuvkou a bude sloužit k nalepení zásuvky ze spodní strany desky.    Podložka navíc může mít větší plochu, než zásuvka, takže bude po přilepení na desce dobře držet.

kaklik, 2019-03-21:
Aktuální návrh vypadá takto

[image]

[image]

Zbývá upravit: 

- [ ]  Názvy proměnných přepsat do angličtiny
- [ ]  Rozměry drážky pro vývod vodičů alespoň 2x žíla 2.5mm² + izolace [příklad vodiče](https://www.tme.eu/cz/details/h07v-k1x2.50bk.100/jednozilove-vodice-lanko/lapp-kabel/4520012s/)
- [ ]  Optimalizovat rozměry pro existující délky šroubů

kaklik, 2019-03-25:
Držák jsem zmenšil v ploše, abych jej stihl vytisknout večer před odjezdem. Pak jsem vyzkoušel jeho montáž. 

[image]

Je super, že držák skutečně pasuje k otvorům v zásuvce.  Pro připojení kontaktů jsem použil úhlové FASTONy, což možná není vhodné řešení. Zároveň ale:

-  Drážka pro vyvedení vodičů od zásuvky je příliš těsná a vodiče se tam nedají ohnout, zároveň by zřejmě bylo vhodné, aby drážka byla ze stejné strany, jako se šroubuje zásuvka, neboť je problém tuhé vodiče udržet v drážce při lepení a zároveň je nepřilepit k podložce, aby bylo možné zásuvku vyměnit
-  Při použití standardních šroubů M3x12 (Nevím pro jakou délku to bylo navrženo, tak jsem použil nejběžnější délku, kterou používáme) jejich konec vyčnívá cca o 1mm nad podložku. Vhodným řešením by tak zřejmě bylo zvýšení tloušťky podložky.

JakubMares96, 2019-03-25:
Ve výkresu od výrobce není uvedená výška uložení šroubu takže to byl odhad. - Předělám. 
Drážku ještě zvětším a asi zapustím níže to by mělo pomoci s uložením vodičů.

kaklik, 2019-03-26:
Já bych tu drážku spíš přehodil na stejnou stranu, co je zásuvka. Bude to mít výhodu, že jak kabeláž tak i zásuvka půjde vyměnit bez sundavání podložky.

kaklik, 2019-04-25:
Vypadá to, že je zde všechno opravené.

closed 2019-04-25 by kaklik