ThunderFly-aerospace/TF-R1 #12: Krabička pro vnitřní zásuvky

kaklik, 2019-03-22:
Krabička může vycházet z [této verze](https://github.com/expediceupice/Raketa/tree/master/3Dprint/igniter), kterou navrhoval původně @ChroustJan 

Akorát je potřeba, aby matice pro upevnění víčka byly zasouvány do kapes z vnější strany krabičky. 

[Zásuvky](https://www.tme.eu/cz/details/procar-57612003/automobilove-napajeci-prislusenstvi/pro-car/57612003/), které mají do krabičky pasovat.

kaklik, 2019-03-27:
Všiml jsem si, že jsi již nahrál soubory s návrhem krabičky. Díval jsem se na ně a je fajn, že jsi to udělalal, mám k nim ale několik připomínek. 

Je potřeba soubory pojmenovat názvy, které neobsahují diakritiku a mezery. Tyto znaky dělají potíže při automatickém zpracování souborů. 

Na samotné krabičce pak vidím tyto potíže

- Krabička nepůjde dobře vytisknout, protože kapsy pro matky nemají  na rozdíl od vzorové krabičky šikmý náběh a tisknou se rovnou do vzduchu. Otvory pro matky jsou konstruovány tak, že matky se musí vkládat z vnitřní strany krabičky. Toto je konstrukční chyba i u vzorové krabičky, neboť matky během šroubování víčka vypadávají. Lepší by bylo je vkládat zvenku. 

Model který vidím vypadá takto:
[image]

Vzorová krabička má šikmý náběh ke kapsám pro matky, právě kvůli usnadnění tisku: 

[image]

-  Do svorky pro vodiče nepůjdou vodiče vložit zboku, protože svorka objímá více než polovinu průřezu vodiče. 
[image]

-  Na gitu je také soubor viko.scad, který ale je efektivně prádný, takže tam zřejmě nemusí být.

JakubMares96, 2019-04-03:
Aktuální podoba:
[image]

[image]

kaklik, 2019-04-16:
Ve skutečném výtisku jsou tyto chyby: 

- Sloupky pro víčko doléhají k modelu jen z jedné strany
- Víčko je jinak tlusté než zbytek krabičky. (tlouška krabičky by měla být sjednocená parametrem wall_thickness)
-  svorky  přívodů předpokládají různý průměr průměr vodiče (průměr na krabičce se liší od průměru na stahovacím dílu)
- Zahloubení pro hlavu šroubu na víčku komplikuje tisk. Podobně i zahloubení na stahovacích svorkách, neboť se pak otvor pro šroub netiskne už od podložky.

roman-dvorak, 2019-07-06:
Model má řadu problémů:

- Otvor na zásuvky je moc úzký. Musí být širokvý 71 mm.
- Rozteče děr na přišroubování desky se zásuvkamy je špatný. Správné rozteče děr jsou 82 x 22 mm
- Otvory pro šrouby skrze panel mají divný rozměr - je potřeba použít správné parametry `M3_screw_diameter`
- Matka se v díře protáčí `M3_nut_diameter`
- V krabičce, kde se mají přišroubovat kabely je potřeba vložit matky do kapsy. Aktuální způsob je velmi složité na složení. A pokud v krabičce budou už namontovány zásuvky, tak je to skoro nesestavitelné.

Na svorce na kabely jsou špatné velikosti válců na kabely.

roman-dvorak, 2019-07-06:
Další problémy jsou, že do víčka nelze zapustit hlavička šroubu - je moc malá díra. To samé platí o svorkách na kabel. 

Pak sloupky pro šrouby v krabičci jsou kus od stěny.

[image]

[image]

[image]

JakubMares96, 2019-07-08:
Tyto problémy vznikly asi tím, že vytištěný model nebyl aktuální nahranou verzí. (již 3 měsíce)
Další věc  se týká zapuštění šroubů. @kaklik, při posledním tisku, na tento problém narazil a měl připomínku "Zahloubení pro hlavu šroubu na víčku komplikuje tisk. Podobně i zahloubení na stahovacích svorkách, neboť se pak otvor pro šroub netiskne už od podložky.". Teď nevím, zda tam to zahloubení dát či ne? 

Do tohoto modelu se má přidat navíc ukazatel napětí a nevím, kam ho umístit. Vedle konektorů, nebo z nějakého boku?  To se asi rozhodne podle toho, jak bude krabička umístěna v autě.

roman-dvorak, 2019-07-08:
Ano, to zahloubení tam má být. Ve víčku určitě. A nad hlavu šroubu vynecháš jednu plnou vrstvu. Stejnak jako jsem Ti psal v chatu o plné vrstvě nad kapsou pro matku. 

Ano, navrhuji posunout otvor pro zásuvky excentricky a otvor pro měřák napětí umístit vedle toho.

roman-dvorak, 2019-07-09:
@JakubMares96 prohlížel jsem si teď ty Tvé modely - ve víčku je určitě špatně nastavená velikost díry pro zapuštění hlavy šroubu. Je nějaký důvodu, že si to počítáš sám (špatně) a nepoužíváš připravený parametr?

Také jsou špatně nastaveny cesty k parametrickému souboru. Proto to vůbec nefunguje.

Jinak např. průměr vodičů, které tam jsou, navrhuji přesunout do parametrického souboru. Aby to  bylo pouze na jednom místě. 

z jakého důvodu kód obsahuje tyto části:

`cylinder(h=M3_nut_height,d=M3_nut_diameter*1.1,$fn=6);`
`M3_nut_diameter*0.9,M3_nut_height*1.2`

Ty parametry jsou nastavené tak, že když se vytisknout, tak jejich hodnota sedí s těmi šrouby.

JakubMares96, 2019-07-12:
Nahrál jsem upravenou verzi. 
+přidání záslepek pro tisk

[image]

[image]

kaklik, 2019-07-17:
Poslední vytištěná krabička má nekonzistentní tloušťku stěn.  Dno je jinak tlusté, než stěny a stěny jsou zřejmě příliš tenké. 
V oblasti víčka jsou dokonce tak, tenké, že slicer tuto část ani netiskne.

Není mi také jasná nutnost tloušťky víčka. Víčko je vzhledem ke krabičce úplně neúměrně tlusté a tiskne se díky tomu asi 8 hodin. 

[image]

roman-dvorak, 2019-07-20:
U krabičky se objevilo několik dalších problémů, které je potřeba vyřešit:

- [ ] Voltmetr se nevejde do otvoru. (obrázky níže) @JakubMares96 můžeš mi prosím poslat odkaz na část kódu, kde se nastavuje velikost otvoru? A kde jsi získal tyto rozměry?

- [ ] Nové krabičce chybí vnější límec - víčko v ní tedy není zapuštěné jako u starší verze. 

### Rozměry voltmetru
[image]

[image]

[image]

### Aktuální stav
[image]

[image]

### Chybějící límec
[image]

[image]

[image]

roman-dvorak, 2019-07-21:
Rozměry celé krabičky lze zmenšit. Délku bych udělal tak, nová délka odpovídala vzdálenosti dvou krajních výztuh. Výška krabičky stačí cca 6-7cm. Výsledkem bude, že se bude tisknout rychleji a v autě nebude zabírat tolik místa.

JakubMares96, 2019-07-22:
Chybějící límec vznikl asi ve sliceru. -> Ozkoušeno nyní by to mělo být OK.
Chyba v otvoru pro voltmer nevím kde vznikla. Tyto rozměry byli nastaveny:
//vyrez pro voltmetr
translate([0,boxwidth/5,-boxheight/2])
cube([45.12,25.9,thicknesswall+5],center=true);

Mohu poprosit o změření rozměru na vytištěném modelu?

kaklik, 2019-07-22:
Vytištěný model nemáme k dispozici.  Tloušťka stěny musí být alespoň 0.61mm jinak se nevytiskne.

roman-dvorak, 2019-07-22:
> //vyrez pro voltmetr
> translate([0,boxwidth/5,-boxheight/2])
> cube([45.12,25.9,thicknesswall+5],center=true);

No, je potřeba k tomu připočítat nějaké tolerance. Odkud vychází to číslo `45.12`? Když jsem naměřil délku `45.08 mm` tak bych nastavil něco klidně jako `45.08 + 1`. Vzhledem k velikosti límce voltmetru není problém přidat i o trochu více např 1.5 mm. 

To samé platí i o šířce `25.8 + 1.5 mm`.

Ten límec bych udělal určitě silnější než 1 vrstvu. Navrhuji `0.8 - 1 mm`. Stěnu takto velké krabičky cca `2 mm`.

JakubMares96, 2019-07-22:
Akorát jsem odevzdal upravenou verzi. Podíváte se, prosím, ve sliceru s Aktuálním nastavením tisku zda je vše OK?

kaklik, 2019-08-01:
Já jako nevím, ale můžeš mi @JakubMares96 vysvětlit, jak když je parametr nastavený takhle:
voltmetr_cutoutwidth = 22.3; 

Tak jak se tam může vejít volmetr, který má šuplerou změřenou šířku 29.19 mm ?

Takováhle práce je úplně k ničemu.

kaklik, 2019-08-04:
Krabička doupravena vrtačkou.

closed 2019-08-04 by kaklik