ThunderFly-aerospace/TF-R1 #14: Základna šestiosé platformy

ChroustJan, 2019-04-17:
Pro písty šestiosé platformy potřebujeme vytvořit držák pístů, který bude určený pro přichycení na střechu a z druhé strany pro přichycení UAV.

Základní platforma by měla vypadat takto:
[image]

Základna musí mít tyto vlastnosti:
-je potřeba jí parametricky umět zvětšovat
-musí být dělitelná na menší díly, aby jí bylo možné tisknout na tiskárně o max délce 15cm
-spojování pomocí šroubů
-horní a spodní základna bude stejná s tím, že na ní budou vložené jiné držáky pístu (bude upřesněno)

kaklik, 2019-04-17:
Horní část základny stejná nebude, neboť je nutné, aby na horní části držela příruba elektropermanentního magnetu. Která se uchycuje 4mi šrouby. Což znamená, že horní díl musí být uzpůsoben k uchycení válce uprostřed. Použití příruby je vidět na této fotografii 

[image]

[Díl na který je příruba uchycena](https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/888_3006.stl)

Taktéž spodní díl by měl být uzpůsobený tomu, že bude ke střeše uchycen na již existující 4 otvory pro šrouby, jejichž vzdálenosti jsou patrné ze stejného dílu, na který je uchycena přiruba.

JakubMares96, 2019-04-18:
[image]

kaklik, 2019-04-18:
Souhlasím, jen pokud horní část základny bude menší než ta spodní, tak bude ještě potřeba, aby ložiska byla nakloněná i dovnitř.

JakubMares96, 2019-04-27:
@kaklik  @ChroustJan 
Návrh nohy pro připojení ložisek[image]

[image]

Středová část
[image]

kaklik, 2019-04-27:
Já ty díly nemůžu na gitu nikde najít. 
Jsou určitě pushnuté?  Hledám je tady: https://github.com/ThunderFly-aerospace/TF-R1/tree/master/CAD/src

Ale co vidím z obrázků: 

- Není mi jasné, proč je tam ten úzký zbytek materiálu mezi ložisky. 
- U připevnění ložisek by mi asi přišlo logičtější, aby otvor na šroub měl ze spodni strany šestihran. Kulové ložisko by se pak k podložce montovalo tak, že se přitáhne maticí ke šroubu, který bude zespodu strčený do podložky. Matice se pak dotáhne klíčem.  V případě jak je to konstruované teďko mi není jasné, jak by bylo možné ten šroub dotáhnout bez sundání celé platformy. 
- Nerozumím funkci otvorů pro šrouby spojující jednotlivé díly. Na společné základně uprostřed chápu, že jsou asi matice. Ale pak mi není jasné, jakým způsobem se dostanou šrouby do dílu s ložisky.

ChroustJan, 2019-05-19:
Horní část základny stejná nebude, neboť je nutné, aby na horní části držela příruba elektropermanentního magnetu. Která se uchycuje 4mi šrouby. Což znamená, že horní díl musí být uzpůsoben k uchycení válce uprostřed. Použití příruby je vidět na této fotografii

[image]

Díl na který je příruba uchycena

Taktéž spodní díl by měl být uzpůsobený tomu, že bude ke střeše uchycen na již existující 4 otvory pro šrouby, jejichž vzdálenosti jsou patrné ze stejného dílu, na který je uchycena přiruba.

ChroustJan, 2019-05-19:
Souhlasím, jen pokud horní část základny bude menší než ta spodní, tak bude ještě potřeba, aby ložiska byla nakloněná i dovnitř.

ChroustJan, 2019-05-19:
Návrh nohy pro připojení ložisek
[image]

[image]
Středová část

[image]

ChroustJan, 2019-05-19:
Já ty díly nemůžu na gitu nikde najít.
Jsou určitě pushnuté? Hledám je tady: https://github.com/ThunderFly-aerospace/TF-R1/tree/master/CAD/src

Ale co vidím z obrázků:

Není mi jasné, proč je tam ten úzký zbytek materiálu mezi ložisky.
U připevnění ložisek by mi asi přišlo logičtější, aby otvor na šroub měl ze spodni strany šestihran. Kulové ložisko by se pak k podložce montovalo tak, že se přitáhne maticí ke šroubu, který bude zespodu strčený do podložky. Matice se pak dotáhne klíčem. V případě jak je to konstruované teďko mi není jasné, jak by bylo možné ten šroub dotáhnout bez sundání celé platformy.
Nerozumím funkci otvorů pro šrouby spojující jednotlivé díly. Na společné základně uprostřed chápu, že jsou asi matice. Ale pak mi není jasné, jakým způsobem se dostanou šrouby do dílu s ložisky.

roman-dvorak, 2019-05-19:
Díly z obrázků jsou v podsložce '[platform](/CAD/src/platform)'. Už mají opravené některé chyby (oproti obrázkům), které popisuješ.

ChroustJan, 2019-05-24:
@JanKott bylo by dobré, kdyby úchytná část pro ložisko byla měnitelným dílem, aby v případě potřeby změnit úhly nebylo potřeba tisknout příliš materiálu.

JanKott, 2019-05-26:
Vytvořil jsem prvotní verzi základny v souborech 888_4003, 4004 a 4006.
Modely includují soubor calculations.scad, který obsahuje výpočty pro získání úhlů pístů a výšku horní platformy. Ze souboru to možná neni uplně jasné, takže můžu v případě potřeby vyrobit nějaký obrázek, kde bude alespoň vidět které trojúhelníky počítám a jak je značím.

kaklik, 2019-05-26:
Vidím, že tam je hodně konstrukčních zlepšení.  

[image]

Zřejmě ale chybí díra, pro matku, která přečnívá z pneumatického válce.  

Mezi ložisky vidím takový podivný útvar, není mi jasné, jak to vzniká. 

[image]

Potom konstrukčně to chápu tak že na prostřední kruhovou část jsou jedním šroubem přišroubovány prodlužovací díly a k těmto dílům přes jeden šroub přišroubovány patky pístů. 

Nejsem si jistý, jestli jeden šroub je dostatečný, pravděpodobně by tam měla být spíš dvojice, aby celé prodloužení nemohlo být zkrouceno okolo jednoho šroubu.  Taktéž mi  nepřijde praktické, aby přístup ke šroubu byl přes celou délku modelu. Protože to pak vyžaduje použití extra dlouhéko inbus klíče. 
Praktičtějši by bylo, kdyby šrouby byly šikmo a přístup k jejich hlavě by byl  šikmo vrchní strany.

JanKott, 2019-05-26:
Díru pro matku pneumatického válce doplnim podle původního modelu (z nějakýho důvodu to je "M6_nut_diameter + 3").

Podivný útvar mezi ložisky je způsoben tim, že ořezávám rameno, kulatou základnu a půlku držáku na písty v půlce. Držák na písty jsem ořízl moc daleko. Myslim, že to nemá to vliv na funkci, ale opravim to.

Šrouby zdvojim a otočim asi až zítra (ale trochu si vymejšlíš :-) ).

JanKott, 2019-05-26:
FYI: abyste to nemuseli hledat - pro upevnění ložiska jsem zvolil šrouby M4 a pro spojování dílů M8x60.

kaklik, 2019-05-26:
Já vim že si vymejšlim, ale šroub M8x60 je něco, co jinak vůbec nikde nepoužijem. Navíc je to předymenzovaný. Myslím ze bez problému bude stačit 2x šroub M4. Třeba 35mm dlouhý.

JanKott, 2019-05-27:
Tak jsem předělal šrouby M8x60 na M4x35, zdvojil jsem je a natočil nahoru.

kaklik, 2019-05-28:
Tak to pasuje :)

[image]

Akorát šrouby pro přišroubování přírubového ložiska k základně by zřemě měly být M5. 
A díra pro přečnívající matku z přírubového ložiska musí být ještě větší, měřením vychází tak na 19.5mm.

JanKott, 2019-05-29:
M4 šrouby jsem vyměnil za M5 a díru na přečnívající matku zvětšil na 19.5mm + global_clearance.

JanKott, 2019-06-01:
Vytvořil jsem dvě knihovny pro zpřehlednění kódu:
1) bolts, která obsahuje modul bolt(size = 4, length = 35, pocket = true, pocket_size = 35, washer = false, square_nut = false)

Čili bez parametru bolt() vyrobí M4x35 šroub s maticí, s přesahem pro šroub a kapsou na matici.
Parametry washer a square_nut zatim nefungují, až je někdo bude potřebovat tak ať si je naimplementuje :-)
Kdo neznal funkci search, doporučuju se podívat na použití tabulky bolts a následného dohledání parametrů šroubu v listu listů.

2) print, která obsahuje balast co normálně dávám na konec modulu print(draft = true, position = [0, 0, 0], rotation = [0, 0, 0])

Je tam hezky vidět použití childů, vezme první child jako model co chci tisknout, druhý a další jako okolí modelu (assembly) a poslední jako výřezovou kostku. Parametry určují pozici a rotaci modelu pro tisk.

čili bez parametru:
print(draft)
{
  model();
  jiny_model();
  cube(10);
}

vytiskne v draftu model, černou barvou průhledně jiny_model a vyřízne z toho kostku 10x10x10.
Pokud je draft = false tak vytiskne jen model v pozici pro tisk.

Pokud jsou childy jen 2, pak nehledá assembly vizualizace.
Pokud je child jen 1, pak nehledá ani výřezovou kostku.

kaklik, 2019-06-06:
Tyhle nový knihovny jsou docela pokrok! :) 

Jinak u té základny je ještě mírná komplikace v tom, že ty šrobouvé spoje mě překvapily v tom, že jak jsou šikmo, tak každý spojený díl je trochu přizvednutý. Takže ta poslední patka pro písty je asi milimetr nad podložkou.  Zatím to ale nevidím jako problém, takže se tohle issue myslím může zavřít. :)

kaklik, 2019-06-10:
Ukázalo se, že platforma zřejmě předpokládá uchycení na střechu takové, že dvě ramena budou symetricky dopředu a jedno rameno dozadu v ose auta. 
Takový předpoklad umístění je správně. 

Potíž ale je, že čtyři otvory šrouby nejsou v horní straně desky vzhledem k osám auta vyvrtány do čtverce, ale do kříže. 

Proto je potřeba otvory v prostředním dílu o 90 stupňů otočit, aby byla možná symetrická montáž.

JanKott, 2019-06-11:
Otvory pro šrouby do desky na autě byly pootočeny o 15 stupňů kvuli konfliktu se šrouby mezi základnou a rameny. Odstranil jsem pootočení (takže teď jde dát platforma buď ramenem rovně dopředu, nebo dozadu). Posunul jsem díry na šrouby mezi základnou a rameny blíže k sobě (aby nebyly opět konfliktní). Díry na šrouby mezi základnou a rameny jsem navíc pootočil ve snaze změnit prohnutí spojených dílů směrem k desce na autě.

Je třeba přetisknout základnu a ramena (stl jsou aktualizované).

kaklik, 2019-06-16:
Nová úprava základny funguje. :)

ChroustJan, 2019-07-02:
Na začátku jsme neočekávali, že bude nutné platformu sundavat. Proto se ukázalo jako lepší řešení, mít šrouby umístěné trvale na střeše vozidla a platformu na ně pouze nandavat a z vrchu dotáhnout křídlovou maticí. Tento fakt vyžaduje změny dílu 888_4004:
https://github.com/ThunderFly-aerospace/TF-R1/blob/24a9beb4cb99a7bc668c879a37c1d554b9eaf6eb/CAD/STL/888_4004.stl

Je potřeba zespodu dílu v místě úchytných šroubů udělat kruhový otvor o průměru 19mm a výšce 10mm.  
[image]

Naopak ze shora je potřeba odstranit otvor na šestihrannou matici.

JanKott, 2019-07-03:
Nakonec jsem kvůli konfliktům se šrouby do ramen nechal 19mm x 2mm prostor pro podložku a nad ní prostor pro M6 matici. Z horní strany jsem odstrani otvor na matici.

closed 2019-07-06 by roman-dvorak