ThunderFly-aerospace/TF-R1 #17: Pohyblivá část šestiosé platformy

kaklik, 2019-05-16:
Pohybující se část šestiosé platformy bude s písty spojená přes kulová ložiska.  
Šrouby těchto ložisek musí být umístěny na kruhové kónickém kuželu. Úhel kónusu  musí odpovídat výšce pístů a poloměru spodní základny #14. 
K uchycení šroubů se předpokládá použití  čtvercových matic. 

Na pohybující části má být umístěn přídržný elektromegnet. 

Díl může vycházet z existujícího dílu https://github.com/ThunderFly-aerospace/TF-G1/blob/master/STL/888_3006.stl

kaklik, 2019-05-19:
Začal jsem s konstrukcí pohyblivého dílu 888_4005

[image]

Stav návrhu tohoto dílu je následující

- [ ] Otvory pro uchycení magnetu
- [ ] Nastavitelnost úhlu ke spodní základně
- [ ] Otvory pro čvercové matice na zašroubování kulových kloubů
- [ ] Zapuštění podložek pro dotažení šroubů kulových kloubů.

kaklik, 2019-06-24:
- Zmenšit výšku otvorů pro čtvercové matice a zajistit, aby měly buď strop a nebo dno.  Aby bylo možné o něco matici opřít před zašroubováním šroubu
- Úchyt pro magnet posunout excentricky co nejvíce dopředu

ChroustJan, 2019-07-02:
Propadla se matice u označené díry fixou na fotce. Nebylo potřeba příliš síly. Prověřit zda tam není málo materiálu.
[image]

roman-dvorak, 2019-07-06:
Za závitem kulového čepu byly vypilovány dvě plošky pro klíč 11 mm. To umožnilo dotáhnout i čepy pod magnetem.

[image]

Na dno děr pro matice je potřeba vytvořit jedna plná vrstva, která zkvalitní tisk. Zároveň může vyřešit problém s propadnutím stěny.

roman-dvorak, 2019-07-07:
Nad matky pro připevnění magnetu byla přidána plná vrstva pro lepší tisk. To by mělo vyřešit možné propadnutí matky. 

Jinak po testu 6. 7. se zdá být tento díl funkční.

closed 2019-07-07 by roman-dvorak