ThunderFly-aerospace/TF-R1 #18: Magnetický držák na telemetrickou antétu

roman-dvorak, 2019-05-26:
K magnetickému držáku pro anemometr s venturiho trubicí je potřeba vytvořit adaptér na telemtrickou anténu.

Adaptér by měl vypadat tak, že to bude takové "víčko", kde na bude vnější závit pro zašroubování do držáku a na dně bude mít díru pro přišroubování antény. Díra musí mít průměr 12 mm.

Existující držák.
https://github.com/MLAB-project/Modules/issues/16

kaklik, 2019-05-26:
@JanKott  Vypadá to, že tvůj magnetický držák bude mít další variantu použití. :)

kaklik, 2019-05-29:
Současné řešení vypadá takhle: 

[image]

To víčko, které na závit toho sloupku pasuje je [tady](https://github.com/MLAB-project/Modules/blob/master/mechanical/WINDGAUGE03A/amf/WINDGAUGE_S01.stl).

roman-dvorak, 2019-06-03:
@cesilko1 vidím nový commit. Je to už tak dodělané? :)

roman-dvorak, 2019-06-06:
Díl je upravený a připravený k výrobě!
 _telemetry_antena_holder.scad_

closed 2019-06-06 by roman-dvorak

kaklik, 2019-06-06:
@cesilko1  Dost by mi usnadnilo práci, kdyby si ty [commity tagoval k issue](https://help.github.com/en/articles/closing-issues-using-keywords). Protože takhle musím ten commit najít a podívat se, jestli si při tom vyrenderoval i tiskový soubor.