ThunderFly-aerospace/TF-R1 #19: Přehrávání zvuku v rádiu z palubního počítače

roman-dvorak, 2019-06-07:
Je potřeba nějakým způsobem dostat zvuk z palubního počitače do rádia v autě. Zvuk by mělo být možné dostat do předního USB hubu (kam je připojen display) kde je, myslím, jack 3 mm výstup. 

Je to potřeba pro hlášeni airspeed z autopilota.

ChroustJan, 2019-06-10:
Myslím, že jsme se koukali a auto nemá vstup jack. Možné řešení by bylo FM vysílač.

roman-dvorak, 2019-06-17:
FM vysílač má výhodu, že telemetrická audio data by bylo možné poslouchat ve všech autech v okolí bez závislosti na jejich (speciálním) vybavení. Tato data jsou užitečná jak pro řidice tážného auta tak pro pilota ale i další zúčastněné.

kaklik, 2019-06-19:
@ChroustJan  myslíš, že bys mohl zkusit najít nějaký FM vysílač, který se ideálně nepřipojuje do cigaretového socketu, ale třeba do USB?

ChroustJan, 2019-06-20:
@kaklik  zkus ještě něco hledat. Ale s tím USB to nevidím dobře. Je to v podstatě produkt do auta.

kaklik, 2019-07-16:
Koupil jsem a otestoval FM vysílač, který problém řeší.

closed 2019-07-16 by kaklik