ThunderFly-aerospace/TF-R1 #20: Telemetrická aplikace pro řidiče a operátora dronu

roman-dvorak, 2019-06-09:
Telemetrická aplikace by měla být velmi jednoduchá a nabízet základní operace pro ovládání bezpilotních dronů. 

Aplikace musí být optimalizovaná pro obrazovky s malým rozlišením a dotykovou plochou. Alternativní ovládání aplikace by mělo být několik tlačítek rozmístěných podél hrany displeje (nebo na jiném, snadno přístupném, místě)

Systém by měl fungovat tak, že bude několik obrazovek (záložek), které se budou moct postupně přepínat. Každá bude obsahovat jiné informace/operace. Na každé obrazovce bude zobrazeo pouze pár údajů tak, aby je bylo možné přečíst velmi rychle bez přerušení pozornosti hledáním dat na obrazovce. 

Měly by existovat obrazovky:
 - Správa dronu - Armování, spuštění kalibrace všech senzorů, 
 - Letové informace - GroundSpeed, AirSpeed, Výška, Vzdálenost, úmělý horizont
 - Stav baterií - podrobné informace o akumulátorech - napětí, 
 - další... 

Ke spojení s dronem by se mělo používat UDP spojení routované z palubního počítače.

roman-dvorak, 2019-06-15:
Při odpojení dronu a následném vypršení timeoutu mavlinku (pymavlink) se knihovna odpojí od UDP a je potřeba toto spojení obnovovat (popřípadě se pokusit o jeho obnovení).

roman-dvorak, 2019-06-17:
Po prvním ostrém používání je potřeba do aplikace přidat několik věcí.

- [ ] Hlášení airspeed se docela osvědčilo. Proto by bylo vhodné aplikaci #9 sloučit s dashboardem. Aby ji bylo možné odtud i ovládat. 

- [ ] Upozornění na vybrané letové režimy. Např. když máme manuální režim, mělo by to průběžně upozorňovat. 

- [ ] Ve spodní liště ukazovat čas od poslední MAVLINK zprávy. Pokud bude tento čas v násobcích 10s, upozornitu na to uživatele a přestat číst všechna data. Ideálně, poomcí nějakých blikátek na to upozornit. 

 - [ ] Po určité době se pokusit o automatické pře-připojení.

- [ ] Vytvořit stránku, která bude určena pro trimování serv. Představoval bych si, že tam bude seznam serv a nad tím tlačitka +, -. Klikáním na ně dojde k přenastavení 'trim' parametru pro každý výstup. 

@ChroustJan Máš nějaké další postřehy a požadavky na tuto aplikace? Zřejmě je pro nás velmi důležitá, takže jakékoliv postřehy z používání jsou vhodné.

ChroustJan, 2019-06-18:
S pohledu samostatného řidiče. Umístění malé obrazovky je nevhodné. Musí být v místě notebooku, nebo přes výdechy větrání palubní desky, ale zároveň nesmí zasahovat do výhledu řidiče. Na daném displeji by nemělo být nic navíc.  V případě, že není spojení musí data zmizet, aby bylo hned jasné, že jsou neplatná. Pokud bude malá obrazovka, měl by být ten sloupec přes co největší plochu. Může tam být  daná odchylka rychlosti. Podle mě bude stačit v jednotkách, bez desetinných míst. Veličina tam být také nemusí. Je to text navíc. Daný sloupec by měl buď respektovat aktuální rychlostní požadavek, aby řidič držel střed ukazatele, nebo aby tam byly vymezeny rychlostní polohy pro jednotlivé módy, táhnutí, nad zemí. Musí tam být signalizace odpojení UAV. Klidně zas tak , že zmizí hodnoty a bude tam nějaký symbol odpojení.

kaklik, 2019-06-18:
A je vůbec vhodné, aby zobrazovaná rychlost byla v m/s, když řidič má před sebou tachometr, který ukazuje v km/h? 

To grafické zobrazení rychlostí by chtělo nějak připravit na zobrazování více veličin zároveň. 
Například by bylo potřeba aby byl zobrazován rozdíl groundspeed vůči airspeed, tak aby z jednoho grafického ukazatele bylo zřejmé, jestli vůbec zvýšením groundspeed jde zvětšit airspeed.  Protože tohle závisí na směru a velikosti větru.  Reálně ale může snadno nastat situace, že potřebná groundspeed by překročila řiditelné limity pro podvozek. Což musí být jasným způsobem poznatelné ze zobrazení.

ChroustJan, 2019-06-19:
Když je to sloupec, tak je to jedno. Řidič nemá čas sledovat tachometr a ještě ten display. 

Řidiči a ostatním by to mělo spíš říci informaci o tom, že start není možný. Dva sloupce jsou již naprd. Může tam blikat nějaký indikátor zeleně (vedle klidně nějaké číslo), že podmínky pro start jsou splněny a červeně, že není možný start. Obecně by to mělo být tak že jakákoliv červená znamená nemožnost startu.

kaklik, 2019-06-19:
Jenže ten indikátor by měl být proporcionální. Neboť start je možný ve velkém rozsahu podmínek, pouze některé jsou větší riziko.  Proto by ten indikátor měl zobrazovat v podstatatě aktuální pozici pracovního bodu v obálce přípustných rychlostí.  Jsou tam ještě totiž další omezení, třeba maximální boční složka větru a podobně.

roman-dvorak, 2019-06-19:
Z těchto zpráv tedy nechápu, jak by měla vypadat telemetrická obrazovka pro řidiče.

Co je jisté:
- Barevný sloupec, ve kterém bude vidět podle barvy, jestli jede rychle nebo pomalu.
- Rychlost v km/h - člověk v autě má přece jen lepší představu o těchto jednotkách
- Ukazovat rozdíl airspeed X groundspeed - možná do sloupcového grafu ukazovat groundspeed pouze čárou přes. A barevný pruh by ukazoval Airspeed 
- Pouze celé jednotky - zbytečné ukazovat desetinná místa

Teď je to udělané tak, že na té obrazovce lze přepnout, jestli se rychlost měří oproti Ground speed nebo Airspeed.

Neudělali byste mi, prosím, nějaký nákres jen na papír? @ChroustJan @kaklik

ChroustJan, 2019-06-20:
Moje představa:
[image]

@roman-dvorak nevidím tam ve tvém přehledu ještě poznámku o detekci odpojení UAV. Zas po odpojení může být červené podbarvení protože pro řidiče je to informace o konci startu.

Jinak až na tu detekci odpojení jsem byl spokojen. Případně nevím zda ještě nejde udělat ukazatel zrušení startu tebou záměrně. Ale nevím na co by jsi jako pilot mačkal.

kaklik, 2019-08-19:
Aplikace aktuálně potřebuje testování, ale nevyžaduje vývojovou práci.

closed 2019-08-19 by kaklik