ThunderFly-aerospace/TF-R1 #23: Překabelování palubního LCD

roman-dvorak, 2019-06-17:
Protože iTec externí dokovací stanice používá pro HDMI nějaký nestandartní driver a rozfugování HDMI na Ubuntu je velmi složité (Může to být způsobeno hromadou dalších věcí - jako nekvalitní USB kabely)

Lenovo dokovací stanice obsahuje 2 grafické výstupy - VGA a DisplaPort (větší varianta). Je potřeba otestovat, jestli by fugovalo řešení DP - HDMI adaptér a cca 2 M HDMI kabelu. Dokovací stanice může pod rádiem zůstat pro napájení LCD displaye a případné připojení nějakého externího zařízení. Např joystick s touchapadem na ovládání displaye.

ChroustJan, 2019-06-20:
@roman-dvorak @kaklik máme 2m HDMI kabel?

ChroustJan, 2019-06-20:
Co tohle: https://www.czc.cz/premiumcord-displayport-na-hdmi-kabel-3m-m-m/107590/produkt?gclid=CjwKCAjwuqfoBRAEEiwAZErCsuYA8Q81aUNjG2vzRIIeWnssioN1tVSdL1iPgU9vKe4HCrL3581O6RoCSEMQAvD_BwE

kaklik, 2019-06-20:
Tohle má dlouhej HDMI konektor, který bude trčet do boku z toho dispeleje.

roman-dvorak, 2019-06-20:
Také by byla varianta pouze sehnat HDMI prodlužovák a DP-HDMI redukci. U
displaye použít ten ohnutý HDMI, co máme.

Jen nevím, jestli ten ohnutý HDMI není na straně u rovného konektoru nějaké
trošku zlomený. Protože když jsem s ním hýbal u sebe v notebooku, tak
display trošku měnil barvy... (Myslel jsem, že tohle v době HDMI nejde a
byla to vlastnost pouze VGA :D )

roman-dvorak, 2019-06-21:
Na tento test jsem našel, že mám:
 - DP - HDMI (F) redukce
 - HDMI - HDMI kabel (asi 2m)
 - DP - HDMI kabel (2 m)

Takže navrhuji to nejdříve zkusit s tímto. A následně se rozhodnout o dalším použití
Přední USB hub bude využit pro napájení LCD a připojení dalších periferií (zvuk, joystick s touchpadem, ...) Zatím se budeme muset vyrovnat s tím, že HDMI kabel bude trčet s displaye. To vyřešíme následně.

a také potřebnou redukci z SMA-R na SMA.

roman-dvorak, 2019-06-29:
LCD display funguje s 2M DP - HDMI kabeleme

closed 2019-06-29 by roman-dvorak