ThunderFly-aerospace/TF-R1 #26: Úprava makefile

kaklik, 2019-07-03:
Makefile by bylo užitečné upravit tak, aby byl schopen renderovat modely bez změny kódu modelů podobně, jako současné řešení u konstrukce WINDGAUGE.

JanKott, 2019-07-04:
Upravil jsem Makefile obdobně jako u WINDGAUGE. Make by teď měl přerenderovat všechny dotčené soubory, včetně závislostí.
POZOR:
Narazil jsem na problém při použití příznaku -j (jobs). Make potom spouští paralelně více MAKE činností, což ovšem může způsobit předbíhání tvorby obrázků před dorenderováním potřebného STL souboru.

roman-dvorak, 2019-07-07:
@JanKott Tato úprava nějak nezafungovala. Nyní je makefile rozbití a při pokusu o renderování dostanu tyto chyby.

```
roman@roman-ultra:~/repos/tf/TF-R1/CAD/src/platform$ make -j 4
make: *** Žádné pravidlo jak vyrobit cíl „/Users/jkt/playground/OpenSCAD/kaklik/TF-R1/CAD/parameters.scad“ potřebný pro „../../STL/assembly.stl“. Konec.

roman@roman-ultra:~/repos/tf/TF-R1/CAD/src/platform$ make 
make: *** Žádné pravidlo jak vyrobit cíl „/Users/jkt/playground/OpenSCAD/kaklik/TF-R1/CAD/parameters.scad“ potřebný pro „../../STL/assembly.stl“. Konec.

roman@roman-ultra:~/repos/tf/TF-R1/CAD/src/platform$ make -W 888_4005.scad 
make: *** Žádné pravidlo jak vyrobit cíl „/Users/jkt/playground/OpenSCAD/kaklik/TF-R1/CAD/parameters.scad“ potřebný pro „../../STL/assembly.stl“. Konec.
```

JanKott, 2019-07-07:
Moje chyba. Nezkontroloval jsem si deps soubory. Problém byl v tom, že na rozdíl od WINDGAUGE nemá TF-R1 všechny závislé zdrojové soubory v adresářích na úrovni modelů, nebo nižších. Konkrétně:
modely jsou v CAD/src/platform
parametry v CAD/
knihovny v CAD/src/lib

odstraňoval jsem tudíž jen cesty k CAD/src/platform, což nestačilo.
Opravil jsem Makefile i deps soubory. Neni to uplně elegantní, ale mělo by to fungovat.

roman-dvorak, 2019-07-07:
Super, díky moc! už to funguje.

closed 2019-07-07 by roman-dvorak

JanKott, 2019-07-08:
Rozdílné cesty navíc způsobily chybné sortování deps souborů. Problém byl, že target pak není na prvním místě a předbíhají ho dependencies. Upravil jsem makefile, ale opět by to chtělo zamyslet se nad nečím elegantnějším.