ThunderFly-aerospace/TF-R1 #27: Připevnit autopilota na střechu auta

roman-dvorak, 2019-07-05:
Na střechu auta by bylo vhodné připevnit autopilota, který bude vyčítat senzory (např. airspeed), bude určovat home-position. Dále z něj můžou být ovládány další věci jako odpojování magnetu pro TF-R1 a start ze střechy, odjišťování ventilů pístů a podobné věci. 

Autopilot by pak měl být přes sériovku připojen do palubního PC, kde bude pomocí mavlink-routered připojen do UDP sítě. Autopilot bude mit vlastní mavlink ID.

kaklik, 2019-07-05:
Musí to být skutečně HW autopilota? Nestačí, když bude pixhawk běžet na palubním počítači s připojenými odpovídajícími senzory?

roman-dvorak, 2019-07-07:
Toto řešení mi příjde také dobré. Jen nevím, co vše to obnáší a kolik to zabere času rozfungovat. Jestli by pak nebylo lepší (rychlejší) naučit se posílat mavlink pakety z pythonu a vyčítat senzory pyMLABem.

kaklik, 2019-07-15:
Myslím, že je vhodnější mít v autě v řetězci co nejméně hardware, který musí fungovat. Takže bych byl asi proto firmware autopilota na střeše auta nahradit software v palubním počítači auta. 

Jedinný závažný argument pro použití hardware autopilota by byl, použít jeho PWM výstupy pro ovládání ventilů platformy.  Na to asi PC se standardnim linuxem asi nebude dost real-time.

kaklik, 2020-03-24:
Pokud máme na měřící systém tyhle požadavky: 

- Získat polohu a orientaci a rychlosti TF-R1 v prostoru
- Umožnit řízení externích komponent
- Publikování dat do lokální sítě
- Možnost relativní navigace UAV oproti autu

Tak z aktuálního průzkumu vyplývají tyhle možnosti.

### Pixhawk autopilot namontovaný na desku auta

Namontovat na střechu auta hardware autopilota (CUAV 5, HKPILOT32), je asi nejpřímější řešení. K palubnímu počítači se hardware autopilota pak musí připojit přes USB/seriový port a nějaký [existující software](https://dev.px4.io/v1.9.0/en/companion_computer/pixhawk_companion.html).

Toto řešení má výhodu v tom, že by z něj mělo být v budoucnu možné pulzně ovládat ventily [šestiosé platformy](https://github.com/ThunderFly-aerospace/6axis-platform).

### IMU snímače připojené k ROS

[ROS](https://en.wikipedia.org/wiki/Robot_Operating_System) umožňuje připojit různé [IMU snímače](https://wiki.ros.org/Sensors#Pose_Estimation_.28GPS.2BAC8-IMU.29) a získat z nich [navigační informace](http://docs.ros.org/melodic/api/robot_localization/html/integrating_gps.html). 

Nevýhodou tohoto řešení je, že by bylo potřeba vyřešit nějaké rychlé a[ realtime připojení SPI senzorů](https://wiki.mlab.cz/doku.php?id=cs:spi#z_usb) k palubnímu počítači. 

### RTK GPS

slimonslimon, 2020-03-26:
Jen komentář ke staršímu postu - pustit PX4 na PCčku a připojit si senzory po I2C přes malb modul jde.

kaklik, 2020-03-26:
> Jen komentář ke staršímu postu - pustit PX4 na PCčku a připojit si senzory po I2C přes malb modul jde.
> 

Ano, to by bylo asi realizovatelné docela snadno. Ale I²C je moc pomalé, ty snímače je potřeba vyčítat tak rychle, aby bylo alespoň trochu možné filtrovat vibrace.

slimonslimon, 2020-03-26:
A SPI je rychlé dost? Možná by se dalo zkusit, zda stejným způsobe není možné pracovat se SPI... Tedy pokud by to k něčemu bylo.

kaklik, 2020-03-26:
> A SPI je rychlé dost? Možná by se dalo zkusit, zda stejným způsobe není možné pracovat se SPI... Tedy pokud by to k něčemu bylo.

Ano s SPI se dá v linuxu pracovat podobně jako s I²C, je ovšem trochu větší problém SPI rozhraní sehnat, nebo vyrobit, protože nejsou device tree drivery do linuxového kernelu třeba na nějaký USB-SPI převodník.  Jsou [tyhle možnosti](https://wiki.mlab.cz/doku.php?id=cs:spi#z_usb).

kaklik, 2020-04-09:
@ChroustJan  na RTK lib se szřejmě dá vykašlat. Ublox podoruje Moving Baseline RTK Configuration. Zkusíme to nakonfigurovat do tohoto módu.  Je to [popsané zde](https://www.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_%28UBX-13003221%29.pdf).

Staci jedna base a dva rovery. Z vektoru se orientace uz spocita.  Pro UAV je to až zbytečně předimenzované. Protože tam zřejmě stačí počítat vektor mezi UAV a TF-R1.