ThunderFly-aerospace/TF-R1 #38: Upevnění hadice k výduchu ventilace auta - Yaris

roman-dvorak, 2019-11-20:
Baterie vírníku je potřeba při teplotach pod 10C vyhřívat. Jeden z možných způsobů je dovést teplý vzduch z ventilace auta pod samotný trup vírníku. 

Na následujících obrázcích je navrhované umístění hadice.

[image]

Hadice (o průměru 75 mm??) by na svém konci měla mít kruhový nástavec, který dosedne na kruhový výduch. 
[image]

Na hadici by o cca 5 cm níže měla být další objímka, která na své spodní straně bude místo pro nalepení suchého zipu. Druhá část zipu bude nalepena v "žlabu" výdechu. 

[image]

Hodice by mezi límcem k výdechu a suchým zpiem měla být zmáčknuta. Tím dojde k domáčknutí límce k výdechu.

kaklik, 2019-12-11:
Tady je trochu nakresleno, jak to @roman-dvorak  myslel. 

[image]

Celá konstrukce jsou dva tištěné kroužky s vnitřním závitem našroubované na hadici. 
První kroužek vlevo má za cíl zarovnat hrdlo hadice tak, aby doléhalo na výstup výdechu auta.  Druhý kroužek našroubovaný na hadici o něco dále slouží k uchycení hadice suchým zipem na palubní desku auta. (jeden kus suchého zipu bude nalepený na kroužku na hadici a druhý kus na palubní desce)

kaklik, 2020-01-08:
@JanKott  začal s tvorbou modelu kroužku pro uchycení hadice.

[image]

Zdrojový soubor jsem [přidal](https://github.com/ThunderFly-aerospace/TF-R1/blob/master/CAD/src/hose_adapter.scad) do repositáře.

roman-dvorak, 2021-02-23:
Tohle bych v aktuálnímu stavu viděl jako nepotřebné. První let s TF-G1 bude na jaře v létě (v teple)

closed 2021-02-23 by roman-dvorak